groeck/it87 #73: NMI on Asus B350M-A

deric, 2018-01-31:
After `modprobe it87` I've starting seeing NMIs:
```
kernel: Uhhuh. NMI received for unknown reason 21 on CPU 1.
kernel: Do you have a strange power saving mode enabled?
kernel: Dazed and confused, but trying to continue
```
the motherboard is B350M-A
```
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: PRIME B350M-A
        Version: Rev X.0x
        Serial Number: 170499493100037
```
`sensors` output:
```
it8655-isa-0290
Adapter: ISA adapter
in0:          +0.87 V  (min =  +0.50 V, max =  +1.08 V)
in1:          +2.50 V  (min =  +0.53 V, max =  +0.99 V)  ALARM
in2:          +1.96 V  (min =  +1.00 V, max =  +0.01 V)  ALARM
in3:          +2.02 V  (min =  +0.46 V, max =  +2.58 V)
in4:          +2.03 V  (min =  +0.50 V, max =  +2.65 V)
in5:          +1.94 V  (min =  +2.50 V, max =  +1.36 V)  ALARM
in6:          +2.01 V  (min =  +2.08 V, max =  +2.02 V)  ALARM
3VSB:         +3.33 V  (min =  +2.42 V, max =  +4.36 V)
Vbat:         +3.27 V  
+3.3V:        +3.33 V  
fan1:        1030 RPM  (min =   45 RPM)
fan2:           0 RPM  (min =   41 RPM)  ALARM
temp1:        +50.0°C  (low  = -53.0°C, high = -81.0°C)  ALARM
temp2:        +46.0°C  (low  = +37.0°C, high = -45.0°C)  ALARM  sensor = thermistor
temp3:        +40.0°C  (low  = -33.0°C, high = +74.0°C)  sensor = thermistor
intrusion0:  ALARM
```

```
$ isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 55 06 44 2d 00 20 c0 00 10 00 69 05 00 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 01 00 38 00 10 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: 00 20 40 00 10 00 00 00 00 20 00 00 10 00 00 00 
d0: ff 00 00 0f 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 10 00 07 07 07 00 00 40 0c 
f0: 00 00 00 00 00 5f 0e 00 00 00 00 00 00 00 00 00
```
```
$ isadump 0x295 0x296
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x295 and data register 0x296.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 01 67 b3 00 00 40 00 00 c0 44 00 00 92 ff 00 
10: 41 e5 00 00 7f 00 00 00 7f 00 00 00 7f 1a 00 00 
20: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 30 c3 99 
30: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 01 01 01 
40: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 00 00 00 
50: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 3c 3c 3c 
60: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 3c 3c 3c 
70: 08 0a 00 4c 4b 00 00 00 00 00 7f 99 00 00 0f 99 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 1b 00 48 
90: 00 00 00 00 00 00 00 ff 40 00 00 00 00 9e 00 00 
a0: 7f 7f 7f 80 00 00 0f 80 7f 7f 7f 80 00 00 0f 80 
b0: 85 a5 00 00 ab f9 27 f9 ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```
might be related to [#60](https://github.com/groeck/it87/issues/60#issuecomment-360356470).

groeck, 2018-01-31:
If you refer to the comments from @cab404, #60 is not a useful problem description. "voltages are all over the place" is likely due to not understanding that voltages need to be scaled, and "temp readings are jumping like crazy" is not actionable information. AFAIK something like the following should work for Asus AB350M-A (from #11, with some fixes).

chip "it8655-*"
    label temp1 "MB CPU Temp"
    label temp2 "Ambient"
    label fan1 "CPU Fan"
    label fan2 "Bottom Fan"
    label fan3 "Back Fan"
    label in0 "Vcore"
    label in2 "+12V"
    label in3 "+5V"
    compute in2 @ + 10, @ - 10
    compute in3 @ * 2.5, @ / 2.5

That doesn't solve the NMI problem, though. No idea where it may be coming from, and you did not provide sufficient context such as the kernel version or boot log. You could try with the ignore_resource_conflict and/or mmio module parameters.

deric, 2018-02-03:
Thanks, I've updated `chip "it8655-*"` configuration. I'm using latest stable kernel from Debian `4.9.65-3+deb9u2`.
```
$ modprobe it87 ignore_resource_conflict=1 mmio=1
$ modinfo it87
filename:       /lib/modules/4.9.0-5-amd64/kernel/drivers/hwmon/it87.ko
version:        v1.0-34-g850c17b
license:        GPL
description:    IT8705F/IT871xF/IT872xF hardware monitoring driver
author:         Chris Gauthron, Jean Delvare <jdelvare@suse.de>
srcversion:     33D9B06F303EC0AC7EAA58F
depends:        hwmon-vid
vermagic:       4.9.0-5-amd64 SMP mod_unload modversions 
parm:           force_id:Override the detected device ID (ushort)
parm:           ignore_resource_conflict:Ignore ACPI resource conflict (bool)
parm:           mmio:Use MMIO if available (bool)
parm:           update_vbat:Update vbat if set else return powerup value (bool)
parm:           fix_pwm_polarity:Force PWM polarity to active high (DANGEROUS) (bool)
```
and here's `sensors` output:
```
sensors
asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

it8655-isa-0290
Adapter: ISA adapter
Vcore:        +0.83 V  (min =  +0.50 V, max =  +1.08 V)  ALARM
in1:          +2.48 V  (min =  +0.53 V, max =  +0.99 V)  ALARM
+12V:        +11.97 V  (min = +11.00 V, max = +10.01 V)  ALARM
+5V:          +5.04 V  (min =  +1.15 V, max =  +6.46 V)
in4:          +2.01 V  (min =  +0.50 V, max =  +2.65 V)
in5:          +1.94 V  (min =  +2.50 V, max =  +1.36 V)  ALARM
in6:          +2.02 V  (min =  +2.08 V, max =  +2.02 V)  ALARM
3VSB:         +3.33 V  (min =  +2.42 V, max =  +4.36 V)
Vbat:         +3.27 V  
+3.3V:        +3.31 V  
CPU Fan:     1046 RPM  (min =   45 RPM)
Bottom Fan:     0 RPM  (min =   41 RPM)  ALARM
MB CPU Temp:  +50.0°C  (low  = -53.0°C, high = -81.0°C)  ALARM
Ambient:      +45.0°C  (low  = +37.0°C, high = -45.0°C)  ALARM  sensor = thermistor
temp3:        +40.0°C  (low  = -33.0°C, high = +74.0°C)  sensor = thermistor
intrusion0:  ALARM
```

groeck, 2018-02-03:
Do you still see NMIs ?

deric, 2018-02-03:
So far none, it seems to be quite random.

groeck, 2018-02-03:
Thanks for the feedback; please add a comment if you see it again.

deric, 2018-02-13:
No, I haven't seen any. I'm gonna reopen the issue is the problem appears again.

closed 2018-02-13 by deric

d235j, 2018-07-06:
I am also experiencing this NMI problem on an Asus TUF B350M-PLUS GAMING board — both with reason 21 and reason 31. It may be related to power settings in the BIOS which I need to double check.

It doesn't seem to affect system operation, though.