groeck/it87 #74: Module loading and unloading weirdness

klausman, 2018-02-14:
Asus Prime B350-PLUS, latest BIOS etc.

I've built from HEAD against the curren 16-rc1 and various version of 4.15.

If I use `insmod ./it87.ko`, that works fine, including good readings and everything.

However, `modprobe it87` does not work after `make modules_install`. There is no error message from `modprobe` or in `dmesg`, but the module just silently not loaded. Furthermore, once the module is loaded, I can't unload it with `rmmod`: I get the message that `it87` is builtin. `rmmod -f it87` works, however.

groeck, 2018-02-14:
Is it possible that the driver _is_ built into the kernel in your system ? Another possibility is that your distribution might use Yet Another Method To Compress Kernel Modules which isn't supported by the Makefile, or yet another location for kernel modules. Can you search for copies of it87.ko* in your system ?

klausman, 2018-02-14:
I don't know if it87 is part of vanilla yet, if it is, it's not obvious.

My kernel is built from a git checkout and there is no distro kernel around at all.

`locate -i it87` only shows my checkout, the modules in `/lib/modules` and unrelated files like `/usr/share/icons/breeze/mimetypes/16/application-x-it87.svg` and `/usr/share/mime/application/x-it87.xml`

groeck, 2018-02-14:
While you are not running a distribution kernel, it is quite likely that the rest of your system is running some distribution, eg Ubuntu or Fedora or whatever. The distribution decides where it wants to look for modules, and it controls if and how those modules are compressed. The distribution also provides a default configuration file, which in turn determines if a driver is built as module or into the kernel. So, even if the kernel is not a distribution kernel, the distribution has a lot of influence.

klausman, 2018-02-14:
Looks like having it87 in the kernel (even if it doesn't support the chip I have) causes the behavior mentioned. Sorry for the noise, I'll be closing this. Is there an ETA on this driver being in-kernel, btw?

closed 2018-02-14 by klausman

groeck, 2018-02-14:
ETA for upstream: None right now, sorry. Upstreaming the code would be a major effort, since I would have to separate all changes into one-patch-per-logical-change chunks. Then there is the problem of reviewers, which are all but impossible to find for hwmon drivers. While it is ok to have un-reviewed code in a repository like this one, that is not acceptable for code in the upstream kernel.
Though, quite frankly, I may at some point make this driver private since I have to spend more and more of my time explaining to people why I don't have time to upstream it - making the situation worse, of course.