groeck/it87 #75: Will not build

LinAGKar, 2018-02-15:
When trying to build this, it complains that the `/lib/modules/4.15.2-1.gb34965a-default/build` is missing. If you create that, it instead complains that it can't find target "modules". It seems that when you run make, it for some reason tries to run `make modules` in `/lib/modules/4.15.2-1.gb34965a-default/build`, which obviously doesn't work since there is nothing there.

groeck, 2018-02-15:
The target directory is derived from "uname -r", which appears to be 4.15.2-1.gb34965a-default in your system. What is your distribution, and what is the expected directory ?

bardacuda, 2018-03-11:
Trying to build this on a fresh install of Manjaro and when I run `make clean` or `sudo make clean` I get the error:

> make[1]: *** No rule to make target 'clean'.  Stop.
> make: *** [Makefile:69: clean] Error 2

groeck, 2018-03-11:
Closing. The original reporter did not respond for three weeks, and the latest report was due to missing headers installation.

closed 2018-03-11 by groeck