groeck/it87 #76: need parameter help with aorus gigabyte gaming 7 motherboard

Goddard, 2018-02-21:
I am trying to get my temprature sensors working, but not really understanding the errors I am getting.  Any help is appreciated.

Here is the dmesg error.

[67684.271451] it87: loading module not compiled with retpoline compiler.
[67684.272589] it87: it87 driver version v1.0-34-g850c17b
[67684.272663] it87: Found IT8686E chip at 0xa40, revision 2
[67684.272701] it87: Beeping is supported
[67684.272721] ACPI Warning: SystemIO range 0x0000000000000A45-0x0000000000000A46 conflicts with OpRegion 0x0000000000000A45-0x0000000000000A46 (\GSA1.SIO1) (20170831/utaddress-247)
[67684.272730] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver

groeck, 2018-02-21:
"loading module not compiled with retpoline compiler" is exactly that. You'll need gcc 7.3.0 or later or accept that the system may be vulnerable against Spectre attacks.
The ACPI warning indicates that the BIOS/ACPI has the memory region used to access the chip reserved for itself. See 'ignore_resource_conflict' module parameter in README, or .acpi_enforce_resources=lax.

groeck, 2018-03-11:
No further response. Closing.

closed 2018-03-11 by groeck