groeck/it87 #77: Support for 0x8738 (Lenovo ThinkCentre Thin Client, M715Q)

stove-panini, 2018-02-25:
This machine is based on AMD's Tiny Bristol Ridge platform, and contains an unknown ITE chip, 0x8738. Here are my results after following your diagnostic procedure:

`# Sensors-detect`
```Probing for Super-I/O at 0x2e/0x2f
...
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8738```

`# isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7`
```     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 
20: 87 38 12 00 00 02 bf 04 c0 10 02 00 8b 04 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 0a 10 0a 00 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 01 20 38 00 04 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: 03 33 04 40 10 00 00 00 02 30 04 00 10 00 0b 98 
d0: ff 00 80 80 00 00 00 00 00 00 00 00 00 00 00 9a 
e0: 00 00 00 00 00 00 00 00 00 20 20 20 c4 3e 3e 0c 
f0: 00 00 00 00 00 00 00 00 00 00 00 00 ff 00 00 00```

Unfortunately, I cannot force an id with modprobe. Attempting to insert it87 at any of the test addresses results in `modprobe: ERROR: could not insert 'it87': Device or resource busy`

Please let me know if I can provide any more information for you. Thanks for all your hard work!

groeck, 2018-02-25:
As usual with ITE chips, there no documentation available anywhere. I don't even know if the chip supports hardware monitoring in the first place. Anyway, if you look in there kernel log (dmesg), you'll probably see that the memory space is reserved by the BIOS/ACPI. Workarounds are available in the README file. 
I don't understand "Attempting to insert it87 at any of the test addresses". What test addresses ? What exactly are you trying to do ?

stove-panini, 2018-02-26:
I was referring to this portion in the README:
```
Next, force-install the driver by providing one of the already supported chips
as forced ID. Useful IDs to test are 0x8622, 0x8628, 0x8728, and 0x8732, though
feel free to test more IDs. For each ID, instantiate the driver as follows
(this example is instantiating driver with ID 0x8622).
	sudo modprobe it87 force_id=0x8622
```
I should've said "ID", not "address".

groeck, 2018-02-26:
Ah, ok. You'll need something like "sudo modprobe it87 force_id=0x8622 ignore_resource_conflict".

stove-panini, 2018-02-26:
That does it, thanks! Values are a little nonsensical, but it's a start. Forcing 0x8628 returns the most data.
```
it8628-isa-0a30
Adapter: ISA adapter
in0:          +0.50 V  (min =  +1.16 V, max =  +2.12 V)  ALARM
in1:          +2.29 V  (min =  +2.89 V, max =  +0.10 V)  ALARM
in2:          +0.24 V  (min =  +1.64 V, max =  +2.94 V)  ALARM
+3.3V:        +0.48 V  (min =  +2.66 V, max =  +1.51 V)  ALARM
in4:          +1.81 V  (min =  +1.74 V, max =  +0.68 V)  ALARM
in5:          +2.71 V  (min =  +2.24 V, max =  +1.99 V)  ALARM
in6:          +1.75 V  (min =  +1.87 V, max =  +3.06 V)  ALARM
3VSB:         +3.60 V  (min =  +3.65 V, max =  +4.30 V)  ALARM
Vbat:         +3.31 V  
+3.3V:        +3.53 V  
fan1:        1046 RPM  (min =   -1 RPM)  ALARM
temp1:        +43.0°C  (low  = +77.0°C, high = +125.0°C)  sensor = thermal diode
temp2:        -70.0°C  (low  = +110.0°C, high = +115.0°C)  sensor = thermal diode
temp3:        +30.0°C  (low  = -97.0°C, high =  +1.0°C)  ALARM  sensor = thermal diode
temp4:        +44.0°C  (low  =  +0.0°C, high =  +0.0°C)  ALARM
temp5:        +12.0°C  (low  = -55.0°C, high = -96.0°C)
temp6:        +60.0°C  (low  = -17.0°C, high = -13.0°C)
intrusion0:  OK
```

groeck, 2018-02-26:
How many fans does the system have ?
Also, can you try a few other models for comparison ? More is not necessarily better (the 3.3V values are wrong, for example - 0x8625 or 0x8732 would give you better results). Do you see temperatures and voltages change under load ? Comparing the 3.3V outputs with the BIOS would show us if 8625 or 8732 is a better fit. Last but not least, the output from "sudo isadump 0xa35 0xa36" would help.

stove-panini, 2018-02-27:
Just one fan.
8625 seems fairly sane in comparison to the rest:
```
it8625-isa-0a30
Adapter: ISA adapter
in0:          +0.46 V  (min =  +1.07 V, max =  +1.95 V)  ALARM
in1:          +2.10 V  (min =  +2.65 V, max =  +0.09 V)  ALARM
in2:          +0.22 V  (min =  +1.51 V, max =  +2.69 V)  ALARM
in3:          +0.22 V  (min =  +1.22 V, max =  +0.69 V)  ALARM
in4:          +1.65 V  (min =  +1.59 V, max =  +0.63 V)  ALARM
in5:          +2.50 V  (min =  +2.06 V, max =  +1.83 V)  ALARM
in6:          +1.61 V  (min =  +1.72 V, max =  +2.81 V)  ALARM
3VSB:         +3.30 V  (min =  +3.34 V, max =  +3.94 V)  ALARM
Vbat:         +3.04 V  
+3.3V:        +3.23 V  
fan1:        1094 RPM  (min =   -1 RPM)  ALARM
fan6:          10 RPM  (min =   -1 RPM)  ALARM
temp1:        +49.0°C  (low  = +77.0°C, high = +125.0°C)  sensor = thermal diode
temp2:        -70.0°C  (low  = +110.0°C, high = +115.0°C)  sensor = thermal diode
temp3:        +32.0°C  (low  = -97.0°C, high =  +1.0°C)  ALARM  sensor = thermal diode
intrusion0:  OK
```
8732, on the other hand, returns incorrect values for +3.3V, but the same temps.

`# isadump 0xa35 0xa36`
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 19 01 ff 0c 1f ff 07 1f ff 87 44 1f 07 66 00 00 
10: 00 00 00 11 c0 00 00 00 02 00 00 00 00 00 00 00 
20: 2a bf 14 14 96 e3 92 96 8a 31 ba 1f 32 0f f3 93 
30: b1 61 08 f1 f5 89 3f 6f 39 91 a6 bb ff 9c b3 98 
40: 7d 4d 73 6e 01 9f 5f 40 ad 6a d4 00 00 00 00 00 
50: ff 07 7f 7f 7f 40 00 00 90 00 37 12 60 00 00 00 
60: 7f 7f 7f 41 00 00 0f ff 7f 7f 7f ff 00 00 0f ff 
70: 7f 7f 7f ff 00 00 0f ff ff ff ff 80 ff ff 0f ff 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 32 00 08 
90: ff 00 00 00 ff 00 00 00 42 99 00 00 01 f5 c0 32 
a0: 3b ea 00 f9 f9 f9 f9 f9 f9 f9 f9 f9 00 00 00 00 
b0: 00 00 00 00 00 00 a0 c9 f3 ef 0f ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
```

groeck, 2018-02-27:
The 8732 code had a bug; that is now fixed.

stove-panini, 2018-02-27:
Pulled the latest and now 8732 reports the following:
```
it8732-isa-0a30
Adapter: ISA adapter
in0:          +0.46 V  (min =  +1.06 V, max =  +1.93 V)  ALARM
in1:          +2.08 V  (min =  +2.63 V, max =  +0.09 V)  ALARM
in2:          +0.22 V  (min =  +1.49 V, max =  +2.67 V)  ALARM
+3.3V:        +0.44 V  (min =  +2.42 V, max =  +1.37 V)  ALARM
in4:          +1.64 V  (min =  +1.58 V, max =  +0.62 V)  ALARM
in5:          +2.47 V  (min =  +2.04 V, max =  +1.81 V)  ALARM
in6:          +1.59 V  (min =  +1.70 V, max =  +2.78 V)  ALARM
3VSB:         +3.27 V  (min =  +3.31 V, max =  +3.90 V)  ALARM
Vbat:         +3.01 V  
fan1:        1035 RPM  (min =   -1 RPM)  ALARM
temp1:        +49.0°C  (low  = +77.0°C, high = +125.0°C)  sensor = thermal diode
temp2:        -70.0°C  (low  = +110.0°C, high = +115.0°C)  sensor = thermal diode
temp3:        +31.0°C  (low  = -97.0°C, high =  +1.0°C)  ALARM  sensor = thermal diode
intrusion0:  OK
```

groeck, 2018-02-27:
Please try again without force_id. Key remaining question is voltage accuracy.

groeck, 2018-02-27:
DMI would never report voltages. Really, the BIOS (like if you boot into the BIOS setup menu) does not report voltages ? Unusual.

stove-panini, 2018-02-27:
That's correct; no voltage is displayed anywhere in the BIOS setup menu.

groeck, 2018-03-11:
Nothing else I can do at this point. Please open another issue if improvements are necessary.

closed 2018-03-11 by groeck