groeck/it87 #78: Report gigabyte hd3p z370

ashwilliams87, 2018-03-03:
Hi!
`# sensors-detect revision 6284 (2015-05-31 14:00:33 +0200)
# System: Gigabyte Technology Co., Ltd. Z370 HD3P [Default string]
# Board: Gigabyte Technology Co., Ltd. Z370 HD3P-CF
# Kernel: 4.15.0-041500-generic x86_64
# Processor: Intel(R) Core(TM) i7-8700 CPU @ 3.20GHz (6/158/10)
`
 I have CPU 4pin cooler and one 120mm 3pin cooler connected to SYS_FAN1 on board.

`Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8686
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
`

` sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 86 02 09 00 00 b3 00 00 00 c0 40 c8 00 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 0a 10 0a 00 0a 20 00 00 00 00 00 00 00 00 00 00 
70: 00 08 20 00 00 06 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 c0 00 00 00 00 
c0: 01 00 00 40 00 ff ff ff 01 00 00 00 00 00 00 00 
d0: e9 00 00 02 00 fd ff ff 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 dc 00 1d 20 12 49 00 00 0a 7e 
f0: 00 00 00 00 00 00 17 00 00 00 49 40 00 ff 00 ff 
`
`sudo isadump -k 0x87,0x01,0x55,0x55 0x4e 0x4f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x4e and data register 0x4f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
10: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
20: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
30: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
40: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
50: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
60: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
70: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
90: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff `

After first installation Ubuntu 16.04 on my PC, sensors did't have any info about RPM, only coretemp.
I added cpi_enforce_resources=lax option to grub.
[... 165 lines cut ...]
Vbat:         +2.90 V  
fan1:         405 RPM  (min =    0 RPM)
fan2:         962 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
temp1:        +30.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +41.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +20.0°C  (low  = +127.0°C, high = +127.0°C)
intrusion0:  ALARM

acpitz-virtual-0
Adapter: Virtual device
temp1:        +16.8°C  (crit = +18.8°C)
temp2:        +27.8°C  (crit = +119.0°C)
temp3:        +29.8°C  (crit = +119.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +32.0°C  (high = +82.0°C, crit = +100.0°C)
Core 0:        +31.0°C  (high = +82.0°C, crit = +100.0°C)
Core 1:        +30.0°C  (high = +82.0°C, crit = +100.0°C)
Core 2:        +28.0°C  (high = +82.0°C, crit = +100.0°C)
Core 3:        +32.0°C  (high = +82.0°C, crit = +100.0°C)
Core 4:        +32.0°C  (high = +82.0°C, crit = +100.0°C)
Core 5:        +30.0°C  (high = +82.0°C, crit = +100.0°C)`

ashwilliams87, 2018-03-03:
Can you help me with that issues

1. fancontrol can't control 3pin cooler, is it actually normal? In BIOS i set temperature config for both coolers, and both have RPM less then 400. But when Ubuntu is booted, 3pin cooler has 900 RPM and it non controlling by fancontrol. As you see:
fan1:         405 RPM  (min =    0 RPM)
fan2:         962 RPM  (min =    0 RPM)
CPU cooler controlling by fancontrol.

2. If I turned off acpi, by added acpi=off to grub, coolers has low RPM. But then i lose HT on my CPU. When i added apci=ht, coolers has maximum RPM.

How to made config when coolers control only by BIOS Fan Smart, not Ubuntu?

groeck, 2018-03-03:
You should install the driver from here and not use the upstream driver. If you do that, you should not need force_id. There is a second chip on on 0x4e/0x4f which you did not report in your sensors-detect output. The correct command for 0x4e/0x4f is "sudo isadump -k 0x87,0x01,0x55,0xaa 0x4e 0x4f 7". You should _not_ use fancontrol but leave fan control to the Super-IO chip. It is built for that and will do a better job. The BIOS lets you set temperature curves; use it. Do not play with ACPI boot flags. Better use the ignore_resource_conflict module option.

ashwilliams87, 2018-03-03:
```
sudo isadump -k 0x87,0x01,0x55,0xaa 0x4e 0x4f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x4e and data register 0x4f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
10: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
20: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
30: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
40: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
50: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
60: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
70: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
90: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
```

ashwilliams87, 2018-03-03:
Thank you groeck! i will try do like you said.

ashwilliams87, 2018-03-08:
Just in case. I found a strange behavior.
1. I set config in BIOS Smart Fan when 3 pin cooler has 300 RPM.
2. System is booting.
3.Sensors: 
fan1: 363 RPM (min = 0 RPM)
fan2:  960 RPM (min = 0 RPM)
fan3: 0 RPM (min = 0 RPM)
4.I sent PC to suspend. 
5.I returned PC from suspend and login to Ubuntu.
6.Sensors:
fan1:         363 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:         269 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)

Is it bug?

groeck, 2018-03-08:
The driver only reports what it gets from the hardware. I can't imagine why it would report different RPMs after boot vs. after suspend/resume. Having said that - if this is reproducible, it might help to provide register dumps for both cases; maybe we can find a configuration register which isn't updated during resume.

ashwilliams87, 2018-03-11:
Groeck thx very much!
I updated bios version my mother board(HD3P) from f4 to f6c and problem has gone.

closed 2018-03-11 by groeck