groeck/it87 #79: cant load it87 - no such device

omen23, 2018-03-20:
Hello,
I am using the debian packages of lm sensor and since I heard on the ryzen platform the nct chip only reads the motherboard temperatures I wanted  to check the CPU temps with it87 - i cloned the git repo and first tried to compile the module without dkms but I cant load it.
$ sudo modprobe it87
modprobe: ERROR: could not insert 'it87': No such device
dmesg output - it87: it87 driver version v1.0-38-g948ad8d but it still doesnt get loaded - why?
modinfo it87 displays the right output and hwmon_vid is loaded ... please help
system: ryzen7 1700x asrock x370 mobo kubuntu 17.10 4.13.0-37-generic x86_64

groeck, 2018-03-20:
ASRock does not commonly use Super-IO chips from ITE; their usual vendor is Nuvoton. Your e-mail suggests that as well by mentioning "nct". What evidence do you have suggesting that this board may use a chip from ITE ? Please at least provide output from sensors-detect.

omen23, 2018-03-20:
I think you are right but the Nuvoton chip does not provide core temperatures, if im right "coretin" is not the core temperature of the processor,
nct6779-isa-0290
Adapter: ISA adapter
Vcore:                  +0.42 V  (min =  +0.00 V, max =  +1.74 V)
in1:                    +1.28 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
AVCC:                   +3.34 V  (min =  +2.98 V, max =  +3.63 V)
+3.3V:                  +3.34 V  (min =  +2.98 V, max =  +3.63 V)
in4:                    +1.85 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                    +0.93 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                    +1.21 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
3VSB:                   +3.46 V  (min =  +2.98 V, max =  +3.63 V)
Vbat:                   +3.28 V  (min =  +2.70 V, max =  +3.63 V)
in9:                    +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in10:                   +0.54 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                   +1.08 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                   +1.66 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                   +0.93 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                   +1.82 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                     0 RPM  (min =    0 RPM)
fan2:                  1401 RPM  (min =    0 RPM)
fan3:                     0 RPM  (min =    0 RPM)
fan4:                     0 RPM  (min =    0 RPM)
fan5:                     0 RPM  (min =    0 RPM)
SYSTIN:                 +35.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
CPUTIN:                 +36.5°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                +15.0°C    sensor = thermistor
AUXTIN1:                +54.0°C    sensor = thermistor
AUXTIN2:                +21.0°C    sensor = thermistor
AUXTIN3:                -25.0°C    sensor = thermistor
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
PCH_MCH_TEMP:            +0.0°C  
intrusion0:            ALARM
intrusion1:            ALARM
beep_enable:           disabled
(sensors output)
Found `Nuvoton NCT5532D/NCT6779D Super IO Sensors'          Success!
    (address 0x290, driver `nct6775')
ok so there is no IT87x chip - sorry for opening the issue - do you have any idea how i can check the real core temperature on this platform? all the best

groeck, 2018-03-20:
Do you have reason to believe that CPUTIN does _not_ reflect the CPU temperature ? Can you share that reason ?
Also, I am confused about your use of "core temperatures". As far as I know, Ryzen only provides a single temperature. If you have different information, please share it with us.
Please note that the k10temp driver (available from github or in the latest kernel) does support Ryzen, though it also only reports a single temperature.

omen23, 2018-03-20:
in the bios the temperature always reads 70°C - there is this confusion about the 20°C offset - so I am wondering which temperature CPUTIN reflects ..  Toms Hardware extract: 
AUXTIN = Auxiliary Temperature Index. 
SYSTIN = System Temperature Index. 
CPUTIN is different from CoreTemp. CoreTemp is the sensor on the processor while CPUTIN is motherboard CPU temp sensor. AUXTIN is the power supply temp sensor if there is one.
best regards

groeck, 2018-03-20:
If the "sensors" command does not show anything after installing k10temp, your kernel version is too old (4.14 or older) and you did not install the driver from https://github.com/groeck/k10temp/.

omen23, 2018-03-20:
Hey!
Thank you - I built the k10temp module from the git repo and it works.
$ sensors
nct6779-isa-0290
Adapter: ISA adapter
Vcore:                  +0.22 V  (min =  +0.00 V, max =  +1.74 V)
in1:                    +1.29 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
AVCC:                   +3.34 V  (min =  +2.98 V, max =  +3.63 V)
+3.3V:                  +3.34 V  (min =  +2.98 V, max =  +3.63 V)
in4:                    +1.85 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                    +0.93 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                    +1.21 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
3VSB:                   +3.46 V  (min =  +2.98 V, max =  +3.63 V)
Vbat:                   +3.28 V  (min =  +2.70 V, max =  +3.63 V)
in9:                    +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in10:                   +0.70 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                   +1.08 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                   +1.66 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                   +0.93 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                   +1.82 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                     0 RPM  (min =    0 RPM)
fan2:                  1406 RPM  (min =    0 RPM)
fan3:                     0 RPM  (min =    0 RPM)
fan4:                     0 RPM  (min =    0 RPM)
fan5:                     0 RPM  (min =    0 RPM)
SYSTIN:                 +35.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
CPUTIN:                 +36.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                +15.0°C    sensor = thermistor
AUXTIN1:                +43.0°C    sensor = thermistor
AUXTIN2:                +21.0°C    sensor = thermistor
AUXTIN3:                -25.0°C    sensor = thermistor
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
PCH_MCH_TEMP:            +0.0°C                                                                                             
intrusion0:            ALARM                                                                                                
intrusion1:            ALARM                                                                                                
beep_enable:           disabled

k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +33.1°C  (high = +70.0°C)
I am still wondering why the BIOS is showing such a high core temperature of 70°C. Thank you for the help, best regards

closed 2018-03-20 by omen23