groeck/it87 #80: temp4 missing from ASUS X399a PRIME

kpande, 2018-04-03:
I'm running the latest `it87` - it works great for PWM control, fan and temperature readings, with the exception that I can't get the PCH temperature.

The BIOS lists 4 temperatures, CPU/MB/VRM/PCH and I noticed yesterday after a day of work, the UEFI showed the PCH at 55C but the rest of the system was a comfortable 30C. I couldn't see this value in Linux (double-checked) and others report being able to view it in Windows (didn't / can't check).

```
[ 1918.178963] it87: it87 driver version v1.0-38-g948ad8d
[ 1918.179035] it87: Found IT8665E chip at 0x290, revision 6
[ 1918.179069] it87: Beeping is supported
```

```
k10temp-pci-00cb
Adapter: PCI adapter
temp1:        +29.8°C  (high = +70.0°C)

it8665-isa-0290
Adapter: ISA adapter
in0:          +0.65 V  (min =  +1.59 V, max =  +0.68 V)
in1:          +0.60 V  (min =  +2.73 V, max =  +1.56 V)
in2:          +2.01 V  (min =  +1.33 V, max =  +1.46 V)
in3:          +2.00 V  (min =  +1.19 V, max =  +0.12 V)
in4:          +0.55 V  (min =  +2.60 V, max =  +1.16 V)
in5:          +0.60 V  (min =  +1.37 V, max =  +0.56 V)
in6:          +0.92 V  (min =  +0.33 V, max =  +1.35 V)
3VSB:         +3.31 V  (min =  +1.81 V, max =  +4.34 V)
Vbat:         +3.05 V  
+3.3V:        +3.33 V  
fan1:        1854 RPM  (min =   81 RPM)
fan2:           0 RPM  (min =   16 RPM)
fan3:           0 RPM  (min =   15 RPM)
fan4:           0 RPM  (min =   -1 RPM)
fan5:           0 RPM  (min =   -1 RPM)
fan6:        2481 RPM  (min =   -1 RPM)
temp1:        +29.0°C  (low  = +98.0°C, high = +120.0°C)
temp2:        +28.0°C  (low  = -79.0°C, high = +63.0°C)  sensor = thermistor
temp3:        +28.0°C  (low  = +73.0°C, high = +60.0°C)  sensor = thermistor
intrusion0:  ALARM

k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +29.8°C  (high = +70.0°C)
```

```
Linux fezzik 4.15.14-gentoo #1 SMP PREEMPT Thu Mar 29 14:52:23 EDT 2018 x86_64 AMD Ryzen Threadripper 1900X 8-Core Processor AuthenticAMD GNU/Linux
```

kpande, 2018-04-05:
also missing with `4.16.0-gentoo`

buraksarac, 2018-06-06:
Also can you please confirm which one is CPU temp? temp1? Here are a bit more info if I can help:

```
dmesg | grep DSDT
[    0.000000] ACPI: DSDT 0x00000000781DD200 00B06C (v02 AMD    A M I    03242016 INTL 20120913)
[    0.079538] ACPI: \_SB_.PCI0.SBRG.EC0_: Used as boot DSDT EC to handle transactions
[    0.099002] ACPI: \_SB_.PCI0.SBRG.EC0_: Used as boot DSDT EC to handle transactions and events
```

```
isadump 0x295 0x296
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x295 and data register 0x296.
Continue? [Y/n] Y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 fe ff bf 00 00 00 00 00 c0 44 88 30 c6 ff ff 
10: 67 d7 ee 70 c0 00 00 00 01 ff ff af 51 9b 00 a8 
20: 3e 3e b7 b7 33 3e 53 98 8f 30 25 2d 2d 2d 30 98 
30: 25 e2 cf 9b 93 66 3e dd fc bf 32 c4 ff ce 22 5c 
40: bb 7d e7 a1 d5 3f b3 ab ad 6a d4 00 5f 40 65 2d 
50: ff 38 7f 7f 7f 40 00 00 90 00 00 12 60 00 00 00 
60: 80 7f 7f ff 00 00 0f ff 80 7f 7f ff 00 00 0f ff 
70: 80 7f 7f ff 00 00 0f ff 80 7f 7f ff 00 00 0f ff 
80: ff ff ff ff 00 00 00 00 00 00 00 00 00 94 00 85 
90: 00 00 00 ff ff 00 00 ff 42 99 00 00 01 2f e0 4b 
a0: 46 46 5f 99 28 00 0f 00 80 7f 7f ff 00 00 0f ff 
b0: fe ce 00 00 f3 d5 cd 3e ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
```

```
isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 65 06 40 3d d0 db 05 04 80 80 49 85 00 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 00 00 02 a0 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 01 00 38 00 10 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: d0 d3 05 04 c0 00 00 00 00 00 04 00 40 00 00 10 
d0: ff 00 00 10 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 10 00 07 07 07 00 00 40 0c 
f0: 00 00 00 00 00 00 0e 00 00 00 00 00 00 00 00 00
```

buraksarac, 2018-06-06:
And here with different chips:

```
it8628-isa-0290
Adapter: ISA adapter
in0:          +0.74 V  (min =  +2.71 V, max =  +0.44 V)
in1:          +0.74 V  (min =  +1.86 V, max =  +2.48 V)
in2:          +2.20 V  (min =  +1.22 V, max =  +1.76 V)
+3.3V:        +4.39 V  (min =  +5.30 V, max =  +1.49 V)
in4:          +0.61 V  (min =  +2.29 V, max =  +3.02 V)
in5:          +0.74 V  (min =  +2.35 V, max =  +0.60 V)
in6:          +1.00 V  (min =  +2.47 V, max =  +3.06 V)
3VSB:         +3.65 V  (min =  +2.21 V, max =  +0.82 V)
Vbat:         +3.43 V  
+3.3V:        +3.65 V  
fan1:        1480 RPM  (min =   15 RPM)
fan2:           0 RPM  (min =   32 RPM)  ALARM
fan3:           0 RPM  (min =   16 RPM)  ALARM
fan5:           0 RPM  (min =   -1 RPM)  ALARM
temp1:        +50.0°C  (low  = +125.0°C, high = -69.0°C)  sensor = thermistor
temp2:        +37.0°C  (low  = -95.0°C, high = -25.0°C)  sensor = thermistor
temp3:        +47.0°C  (low  = +63.0°C, high = -43.0°C)  sensor = thermistor
temp4:        +47.0°C  (low  = -43.0°C, high = -13.0°C)
temp5:        +47.0°C  (low  = +62.0°C, high = -51.0°C)
temp6:        +51.0°C  (low  =  -1.0°C, high =  -1.0°C)
intrusion0:  ALARM

k10temp-pci-00cb
Adapter: PCI adapter
temp1:        +50.0°C  (high = +70.0°C)

amdgpu-pci-4200
Adapter: PCI adapter
vddgfx:       +0.85 V  
fan1:         971 RPM
temp1:        +35.0°C  (crit = +79.0°C, hyst = -273.1°C)
power1:       20.00 W  (cap = 300.00 W)

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +50.0°C  (high = +70.0°C)

---------------
it8607-isa-0290
Adapter: ISA adapter
in0:          +0.74 V  (min =  +2.71 V, max =  +0.44 V)  ALARM
in1:          +0.74 V  (min =  +1.86 V, max =  +2.48 V)  ALARM
in2:          +2.20 V  (min =  +1.22 V, max =  +1.76 V)  ALARM
in3:          +2.20 V  (min =  +2.65 V, max =  +0.74 V)  ALARM
in4:          +0.61 V  (min =  +2.29 V, max =  +3.02 V)  ALARM
in5:          +0.74 V  (min =  +2.35 V, max =  +0.60 V)  ALARM
in6:          +1.00 V  (min =  +2.47 V, max =  +3.06 V)  ALARM
3VSB:         +3.65 V  (min =  +2.21 V, max =  +0.82 V)  ALARM
Vbat:         +3.43 V  
+3.3V:        +3.65 V  
fan2:           0 RPM  (min =   32 RPM)  ALARM
fan3:           0 RPM  (min =   16 RPM)  ALARM
temp1:        +49.0°C  (low  = +125.0°C, high = -69.0°C)  ALARM  sensor = thermistor
temp2:        +37.0°C  (low  = -95.0°C, high = -25.0°C)  ALARM  sensor = thermistor
temp3:        +46.0°C  (low  = +63.0°C, high = -43.0°C)  ALARM  sensor = thermistor
intrusion0:  ALARM

k10temp-pci-00cb
Adapter: PCI adapter
temp1:        +49.2°C  (high = +70.0°C)

[... 187 lines cut ...]
fan3:           0 RPM  (min =   16 RPM)  ALARM
temp1:        +49.0°C  (low  = +125.0°C, high = -69.0°C)  ALARM  sensor = thermistor
temp2:        +37.0°C  (low  = -95.0°C, high = -25.0°C)  ALARM  sensor = thermistor
temp3:        +45.0°C  (low  = +63.0°C, high = -43.0°C)  ALARM  sensor = thermistor
intrusion0:  ALARM

k10temp-pci-00cb
Adapter: PCI adapter
temp1:        +49.0°C  (high = +70.0°C)

amdgpu-pci-4200
Adapter: PCI adapter
vddgfx:       +0.85 V  
fan1:         972 RPM
temp1:        +35.0°C  (crit = +79.0°C, hyst = -273.1°C)
power1:       20.00 W  (cap = 300.00 W)

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +49.0°C  (high = +70.0°C)
```

buraksarac, 2018-06-06:
sensors-detected reported 0x4e/0x4f but i got only ff
```
Found `ITE IT8665E Super IO Sensors'                        Success!
    (address 0x290, driver `to-be-written')
Probing for Super-I/O at 0x4e/0x4f

isadump -k 0x87,0x01,0x55,0x55 0x4e 0x4f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x4e and data register 0x4f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
10: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
20: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
30: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
40: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
50: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
60: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
70: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
90: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
```

groeck, 2018-06-06:
0x4e/0x4f needs the following command.
`isadump -k 0x87,0x01,0x55,0xaa 0x4e 0x4f 7`
I'll need to update README to reflect this.

groeck, 2018-06-06:
Please try the latest version of the driver.

kpande, 2018-06-06:
```
temp1:        +40.0°C  (low  = +115.0°C, high = +120.0°C)
temp2:        +30.0°C  (low  = -79.0°C, high = +63.0°C)  sensor = thermistor
temp3:        +30.0°C  (low  = +73.0°C, high = +60.0°C)  sensor = thermistor
temp4:        +30.0°C  (low  = -81.0°C, high = +46.0°C)  sensor = thermistor
temp5:        +30.0°C  (low  =  -1.0°C, high = -21.0°C)  sensor = thermistor
temp6:        +31.0°C  (low  = -84.0°C, high = +121.0°C)  sensor = thermistor

```

I don't know what the new values mean, but they show up. not sure if they are legitimate.

buraksarac, 2018-06-07:
Yes i can see same result as @kpande 

```
it8665-isa-0290
Adapter: ISA adapter
in0:          +0.68 V  (min =  +2.46 V, max =  +0.40 V)
in1:          +0.68 V  (min =  +1.69 V, max =  +2.26 V)
in2:          +2.00 V  (min =  +1.11 V, max =  +1.60 V)
in3:          +2.00 V  (min =  +2.41 V, max =  +0.68 V)
in4:          +0.56 V  (min =  +2.08 V, max =  +2.75 V)
in5:          +0.68 V  (min =  +2.14 V, max =  +0.55 V)
in6:          +0.91 V  (min =  +2.25 V, max =  +2.78 V)
3VSB:         +3.29 V  (min =  +2.01 V, max =  +0.74 V)
Vbat:         +3.14 V  
+3.3V:        +3.31 V  
fan1:        1548 RPM  (min =   15 RPM)
fan2:           0 RPM  (min =   32 RPM)
fan3:           0 RPM  (min =   16 RPM)
fan4:           0 RPM  (min =   -1 RPM)
fan5:           0 RPM  (min =   -1 RPM)
fan6:           0 RPM  (min =   -1 RPM)
temp1:        +45.0°C  (low  = +125.0°C, high = -69.0°C)
temp2:        +32.0°C  (low  = -95.0°C, high = -25.0°C)  sensor = thermistor
temp3:        +40.0°C  (low  = +63.0°C, high = -43.0°C)  sensor = thermistor
temp4:        +40.0°C  (low  = -85.0°C, high = -77.0°C)  sensor = thermistor
temp5:        +40.0°C  (low  = -43.0°C, high = -13.0°C)  sensor = thermistor
temp6:        +40.0°C  (low  = +62.0°C, high = -51.0°C)  sensor = thermistor
intrusion0:  ALARM

k10temp-pci-00cb
Adapter: PCI adapter
temp1:        +45.0°C  (high = +70.0°C)

amdgpu-pci-4200
Adapter: PCI adapter
vddgfx:       +0.94 V  
fan1:         966 RPM
temp1:        +32.0°C  (crit = +79.0°C, hyst = -273.1°C)
power1:       19.00 W  (cap = 300.00 W)

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +45.0°C  (high = +70.0°C)

```
isadump still same:
```
isadump -k 0x87,0x01,0x55,0xaa 0x4e 0x4f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x4e and data register 0x4f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
10: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
20: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
30: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
40: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
50: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
60: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
70: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
90: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
```

groeck, 2018-07-09:
Closing. Please see driver removal notice in README file.

closed 2018-07-09 by groeck