groeck/it87 #82: Asus ROG Strix X370-I

dtltf, 2018-04-29:
Hi,

I have **Asus Rog Strix X370-I** keyboard. No sensors work. Using `sensors-detect` gives me this (as in Readme)

Here are the outputs

```
    Probing for Super-I/O at 0x2e/0x2f
    ...
    Trying family `ITE'...                                      Yes
    Found unknown chip with ID 0x8665
	(logical device 4 has address 0x290, could be sensors)
```

Here are the outputs

`sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7`

```
0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 65 06 40 3d 50 ff cf 22 95 00 49 85 08 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 01 00 38 00 10 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: 50 ff cd 22 d5 00 00 00 40 00 00 00 10 08 00 00 
d0: ff 00 00 0c 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 10 00 07 07 07 00 00 40 0c 
f0: 00 00 00 00 00 19 0e 00 00 00 00 00 00 00 00 00 
``` 

`sudo isadump 0x295 0x296`

```
0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 32 ff bf 00 00 03 00 00 c0 44 08 00 e6 ff 00 
10: b7 c6 b3 30 c0 80 80 00 01 ff 00 f6 53 56 00 00 
20: 51 e5 b6 b8 03 03 03 97 95 29 2b 2b 2b 2b 2b 98 
30: e8 b5 1e 00 b8 d0 fc e6 72 ad 64 de 2b 53 55 64 
40: ef e6 b0 6b 22 06 fd ba ad 6a d4 00 5f 40 c4 2f 
50: ff 38 55 7f 7f 40 00 00 90 00 00 12 60 00 00 00 
60: 80 14 46 33 16 00 0f 6c 80 1e 46 99 14 00 0f b4 
70: 7f 7f 7f 99 00 00 0f 99 7f 7f 7f 99 00 00 0f 99 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 8f 00 49 
90: 00 00 00 ff ff 00 00 ff 42 99 00 00 01 0b e0 29 
a0: 7f 7f 7f 80 00 00 0f 80 80 7f 7f ff 00 00 0f ff 
b0: 4a 24 00 00 91 2a f6 fa ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
```

Please provide some more instructions on what can I do.

groeck, 2018-04-29:
If you have the driver from here installed, most likely you have an ACPI resource conflict. Please check dmesg output.

dtltf, 2018-04-29:
Hello!

Thank you for the answer and your time. I really appreciate it.

So I reinstalled the driver and run `sensors-detect` again. While it gave me the same output, `sensors` actually detected something. I'm pasting the results here:

```
asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

amdgpu-pci-0900
Adapter: PCI adapter
fan1:         791 RPM
temp1:        +27.0°C  (crit =  +0.0°C, hyst =  +0.0°C)

it8665-isa-0290
Adapter: ISA adapter
in0:          +1.11 V  (min =  +0.58 V, max =  +2.53 V)  ALARM
in1:          +2.50 V  (min =  +0.00 V, max =  +0.33 V)  ALARM
in2:          +1.98 V  (min =  +2.27 V, max =  +2.01 V)  ALARM
in3:          +2.01 V  (min =  +2.51 V, max =  +2.76 V)  ALARM
in4:          +0.03 V  (min =  +1.89 V, max =  +1.24 V)  ALARM
in5:          +0.03 V  (min =  +2.42 V, max =  +1.09 V)  ALARM
in6:          +0.03 V  (min =  +0.91 V, max =  +0.47 V)  ALARM
3VSB:         +3.31 V  (min =  +2.18 V, max =  +1.85 V)  ALARM
Vbat:         +3.23 V  
+3.3V:        +3.31 V  
fan1:        1134 RPM  (min =   10 RPM)
fan2:           0 RPM  (min =   28 RPM)  ALARM
fan6:           0 RPM  (min =   -1 RPM)
temp1:        +34.0°C  (low  = -90.0°C, high = -17.0°C)  ALARM
temp2:        +38.0°C  (low  = +107.0°C, high = -16.0°C)  ALARM  sensor = thermistor
temp3:        +37.0°C  (low  =  +6.0°C, high = +34.0°C)  ALARM  sensor = thermistor
intrusion0:  ALARM

k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +83.8°C  (high = +70.0°C)
```

While it's a step forward, it looks like the data is inaccurate or I can't seems to grasp it. It's impossible that the pci adapter is 83 degrees. While measuring temps on Windows, my CPU sits around ~50 degrees while the GPU is around ~40.

Output of `dmesg | grep "it87"`

```
[ 1064.580235] it87: loading out-of-tree module taints kernel.
[ 1064.580950] it87: it87 driver version <not provided>
[ 1064.581018] it87: Found IT8665E chip at 0x290, revision 6
[ 1064.581052] it87: Beeping is supported
```

What can I do to get actual sensors? Any help will be much appreciated. I can get my hands dirty if that could help, although it's obvious I'm rather new to this sorts of things. Thank you for your work!

dtltf, 2018-04-30:
Hello,

Thanks for the answer! Sorry I didn't notice the last paragraph, my mistake.

Regarding `k10temp`, I have AMD Ryzen 7 2700. My kernel is 

`4.16.5-1-ARCH`

I tried installing `k10temp` but it failed due to `System.map` file not present. Do I need to compile my own kernel for it to work? Or is there another way?

groeck, 2018-05-05:
Yes, you'll have to adjust Makefile. Try again. Untested; I pulled the changes from a recent commit into the it87 driver repository.

groeck, 2018-05-10:
Closing. If problem still persists please add a comment and provide details.

closed 2018-05-10 by groeck