groeck/it87 #84: Having trouble building on 4.16.6

m00dawg, 2018-05-02:
I have to imagine I'm doing something wrong but I can't seem to get it87 built when using 4.16.6. Previous kernels I could just go into the it87 project and run a 'make' and 'make install' but this time around I get:

```
tim@lindawg ~/git/it87 $ make
  CC [M]  /nfs/home/tim/git/it87/it87.o
/bin/sh: 1: ./tools/objtool/objtool: not found
scripts/Makefile.build:330: recipe for target '/nfs/home/tim/git/it87/it87.o' failed
make[2]: *** [/nfs/home/tim/git/it87/it87.o] Error 127
Makefile:1561: recipe for target '_module_/nfs/home/tim/git/it87' failed
make[1]: *** [_module_/nfs/home/tim/git/it87] Error 2
Makefile:71: recipe for target 'modules' failed
make: *** [modules] Error 2
```

I thought it was because I'm not building a deb package during the build or something but my previous kernel (4.13.10) doesn't seem to have that issue using the same build scripts.

Trying out 4.16.6 since I've upgraded to a Ryzen 2700X and was trying to sort out the temperature differences I seem to be getting again (plus 4.16.6 I think fixes a few issues with 2700X if I'm not mistaken)

groeck, 2018-05-02:
The kernel build now requires objtool to be installed in the headers file, but the scripts to generate header packages don't include it. My workaround is to manually copy objtool from my local build directory into the headers directory. Something like
    sudo cp tools/objtool/objtool /usr/src/linux-headers-4.16.6/tools/objtool/objtool
No idea how to fix the problem for good, sorry.

m00dawg, 2018-05-02:
Ah man I thought it was something like that but hadn't quite figured it out. Thanks for the nudge in the right direction. After doing that I was able to build the module without issues. Thanks!

closed 2018-05-02 by m00dawg