groeck/it87 #85: module won't install due to error

KeithMyers, 2018-05-08:
Just installed the 4.15.0-20 kernel and went to make the it87 module.  Everything looked normal but when I lsmod'd, it wasn't loaded.  So I did a modprobe it87 and it throws an error now.  The module was loading through all the previous kernels I have used.  This is the error message.

root@Darksider:/home/keith/Documents/it87-master# modprobe it87
modprobe: ERROR: could not insert 'it87': Unknown symbol in module, or unknown parameter (see dmesg)

This the dmesg error.

[  299.517047] it87: Unknown symbol vid_from_reg (err 0)
[  299.517071] it87: Unknown symbol vid_which_vrm (err 0)

So has the module not been checked to function yet with this kernel?  What can I do to get it to compile and install normally with this kernel?

groeck, 2018-05-08:
The driver requires CONFIG_HWMON_VID to be enabled. This has always been the case. The question is why this configuration option is disabled in your kernel.

KeithMyers, 2018-05-08:
OK, thanks for the reply.  So where is the best place to ask why that configuration isn't enabled for that kernel?

KeithMyers, 2018-05-08:
OK, I figured the kernel installation wasn't complete or something, so booted the previous kernel and purged the 4.15.0-20 installation.  Then tried again.  This time the installation picked up the it87.ko driver as usual since it is set up in the dpkg configuration.  Everything back to normal with the driver picking up my Ryzen mobo sensors.  Thanks for setting me straight.

closed 2018-05-08 by KeithMyers