groeck/it87 #89: Most fan outputs missing from ASUS ROG Crosshair VII Hero with the driver

KeithMyers, 2018-05-26:
I hope that the driver interaction with the Crosshair VII Hero motherboard can be investigated.
Only Fan1 and Fan5 are reported.  Only Fan1 actually reports a value.  Fan5 never reports anything.
All the other fan header outputs are completely missing.  I have tried all fan control settings in the BIOS and none make any difference.

I would like to know why the same it87 driver reports all 6 fan headers on the ASUS X370 Prime Pro with the same ITE8665E SIO chip and doesn't with the C7H.

keith@Serenity:~$ sensors
asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

it8665-isa-0290
Adapter: ISA adapter
in0:          +1.29 V  (min =  +2.37 V, max =  +2.74 V)
in1:          +1.40 V  (min =  +2.67 V, max =  +2.39 V)
in2:          +2.40 V  (min =  +2.73 V, max =  +0.72 V)
in3:          +1.97 V  (min =  +2.09 V, max =  +2.66 V)
in4:          +1.11 V  (min =  +1.16 V, max =  +1.69 V)
in5:          +0.50 V  (min =  +2.52 V, max =  +1.80 V)
in6:          +0.97 V  (min =  +2.43 V, max =  +2.78 V)
3VSB:         +3.36 V  (min =  +2.51 V, max =  +5.38 V)
Vbat:         +3.27 V  
+3.3V:        +3.36 V  
fan1:        2472 RPM  (min =   10 RPM)
fan5:           0 RPM  (min =   -1 RPM)  ALARM
temp1:        +58.0°C  (low  =  -5.0°C, high = +93.0°C)
temp2:        +30.0°C  (low  =  -5.0°C, high = -42.0°C)  sensor = thermistor
temp3:        +44.0°C  (low  = -113.0°C, high = -69.0°C)  sensor = thermistor
intrusion0:  ALARM

k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +68.1°C  (high = +70.0°C)
Tctl:         +78.1°C  

keith@Serenity:~$

groeck, 2018-05-26:
I'll need a register dump (both sets of registers) to have a chance to track this down. Also, please provide relevant output from dmesg and/or sensors.conf in case there is a second (unsupported) Super-IO chip on the board.
Note that an online review states "We find a ROG chip that seems to expand fan control and other ROG features and a TPU chip that should enhance monitoring and OC features". This suggests that there may be an undocumented/proprietary chip handling the additional fans.

KeithMyers, 2018-05-28:
I'll need some direction to  get you core dumps.  Couldn't find anything in dmesg other than the it87 beep enabled message.  No sensors.conf file.

I was mistaken, the Fan5 sensor output is the H_Amp header on the motherboard and reports normally if a fan is attached to it.

Yes, there is a ROG chip on the motherboard but I am unsure of its purpose.  All the fan headers show up in the BIOS under the Q-Fan control so that seems to indicate something is riding shepherd on the control of them.

KeithMyers, 2018-06-25:
I think I figured out to make the dump you asked for.
sudo isadump 0x295 0x296
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x295 and data register 0x296.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 fa 7f be 00 00 00 00 00 c0 44 08 30 63 ff ea 
10: ef 55 ac 70 c0 00 00 00 01 ff 01 fd 5e ed 00 00 
20: 76 80 db b4 66 38 58 99 96 46 22 30 30 30 30 9a 
30: fb f9 df f5 4a fa f4 c2 99 6a ad e7 bf df f7 77 
40: 5d db f6 fb bb 8f fb e5 ad 6a d4 00 5f 40 db ff 
50: ff 38 7f 7f 7f 40 00 00 90 00 00 12 60 00 00 00 
60: 80 7f 7f ff 00 00 0f ff 80 7f 7f ff 00 00 0f ff 
70: 80 7f 7f ff 00 00 0f ff 80 7f 7f ff 00 00 0f ff 
80: ff ff 26 02 00 00 00 00 00 00 00 00 00 91 00 25 
90: 00 00 00 ff ff 00 00 ff 42 99 00 00 01 cd e0 50 
a0: 80 7f 7f ff 00 00 0f ff 80 7f 7f ff 00 00 0f ff 
b0: 40 7e 00 00 d5 97 96 4d ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff

adatum, 2018-07-09:
Any progress on this? I have the same motherboard and it would be great to see more sensor details and especially fan speeds if it's possible.

groeck, 2018-07-09:
The register dump provided earlier is incomplete and does not include the SIO registers. FWIW, it does suggest that only two fans are connected to the chip (fan1 and fan5). Without SIO register dump I have no means to determine if the other fans are disabled or just not connected.

groeck, 2018-07-09:
Please note: If fan control for the additional fans is indeed handled by the ROG chip, there is nothing we can do unless Asus releases specifications of this chip. Even then, someone with an affected board would have to be willing and have the time to write a driver.

adatum, 2018-07-09:
Do you mean my dump is incomplete or that of @KeithMyers ? Let me know what else you need from me.

I see only fan1 (CPU_FAN) and fan5 (H_AMP) ever listed by `sensors`, but I also have fans connected to CPU_OPT (mirror of fan1), CHA_FAN1, and CHA_FAN2. See [page 1-19 (or 35 of 108 of the PDF) in the manual](http://dlcdnet.asus.com/pub/ASUS/mb/SocketAM4/ROG_CROSSHAIR-VII-HERO_WI-FI/E13834_ROG_CROSSHAIR_VII_HERO_WI-FI_UM_WEB.pdf) for the fan headers corresponding to the labels. 

[image]

groeck, 2018-07-09:
Both, really. The SIO register dump would have to be 
    sudo isadump -k 0x87,0x01,0x55,0xaa 0x4e 0x4f 7
Notice the 0xaa instead of 0x55. I would also need
    sudo isadump -k 0x87,0x01,0x55,0xaa 0x4e 0x4f 0
    sudo isadump -k 0x87,0x01,0x55,0xaa 0x4e 0x4f 4
to be sure I catch everything.

KeithMyers, 2018-07-09:
I can do the same dump if it will show anything new.  I don't think that will be the case.  I didn't know how to get the SIO registers until you posted the instructions.

I am working with the System Information Viewer utility developer Ray Hinchcliffe to identify the fan headers for this board in the Windows environment.  He has always been able to develop new versions of his program that fully report all motherboard fan headers for each new product on the market.  He already has got the AIO cooler to report as well as CPU Fan, CPU Opt, Fan1, Fan2, Fan3 and W_flow headers.  I will get him the rest of the fan headers tomorrow when I fully populate every fan header on the board.

My question is . . . . what makes it so easy to figure out motherboard fan headers in Windows that are so difficult or impossible in Linux?  What tools are missing or not available?  All his program needs to do is export two files for him to figure out any motherboard.  One is a text file and the other is a DMI file.

Would either of those files be useful to you?  I can attach them to this thread if any are useful.

groeck, 2018-07-09:
Problem is that the Super-IO in your system is at 0x2e/0x2f, not at 0x4e/0x4f. With that, the correct commands would be

```
sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 0
sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 4
sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
```

As for development under Windows, you'll have to ask Mr. Hinchcliffe how he gets the information he needs. Other Windows tools developers sign NDAs with various companies. This won't work with Linux, since the result is open source, and we would violate an NDA by publishing source code created with information obtained from it. Also, most of the time Linux developers work for companies and can not legally sign any NDA without violating company policies. Windows tools developers do not have that problem since they only deliver binary code, not source. On top of that, board manufacturers such as ASUS officially support Windows, but explicitly do _not_ support Linux. They are willing to give more information to Windows tools developers than they are willing to give to Linux developers. We (Linux developers) sometimes do have some contacts with board and chip companies, but those contacts are always at risk of drying up.
In addition to all that, Windows tools developers usually make money one way or another with their tools. They can spend a lot of time figuring out details; they have to do this to stay in the market. Are you aware that I am _not_ getting paid a dime to do this work, that it is all on my own time, time taken away from my family ? I guess not, or there would not be all those demands for free development and support. Are _you_ working for free, or do you just demand free support from others ?

KeithMyers, 2018-07-09:
I see that adatum has beaten me to the relevant needed SIO dumps you asked for.  I have been using SIV for probably ten years now.  The developer has never asked for a dime from any of his users since the software is freeware.  I'm sure he is employed as a developer but the SIV program is not the source of his income.  Some of the best utilities I use are freeware and usually a much better product than any commercial solution.

I am not aware of any special information that Ray receives from manufacturers but I do think he is the beneficiary of early product releases.  Maybe he gets paid by the manufacturers for debugging his program on new hardware. But if that is the case I wouldn't expect his software to be free and published by him solely and not the manufacturers intellectual property.  I don't know how all of the business of this works and I am just postulating.

Thanks for pointing out the differences in developing for LInux versus Windows.

groeck, 2018-07-09:
Closing as WontFix. Please see driver removal notice in README file.

closed 2018-07-09 by groeck

KeithMyers, 2018-07-09:
I apologize if I have offended you.  I do appreciate your efforts.  I am very sorry you are removing the driver.