groeck/it87 #90: determining for which kernel the modules is built

udovdh, 2018-05-27:
How can one control for which kernel the module is built?
For the running kernel?
For the kernel in /usr/src/linux?\
For the latest installed kernel?
I.o.w.: why does it copy a module for e.g. 4.16.11 into /lib/modules/4.16.9 on a `make install`?

groeck, 2018-05-27:
It builds for the currently running kernel. If you want to build for another kernel, you'll have to modify Makefile accordingly.

udovdh, 2018-05-30:
Thanks.
in 4.16.13 it sees this:
# dmesg|grep it87
[   54.729998] it87: Found IT8733E chip at 0xa60, revision 3
[   54.734722] it87: Beeping is supported
[  162.390334] it87: loading out-of-tree module taints kernel.
[  162.390836] it87: it87 driver version v1.0-42-g7182608
[  162.390913] it87: Found IT8686E chip at 0xa40, revision 2
[  162.390968] it87: Beeping is supported
[  162.392614] it87: Found IT8792E/IT8795E chip at 0xa60, revision 3
[  162.392659] it87: Beeping is supported

First is what the kernel's it87 sees.
Then I build he module and install it.
I remove it87 and modprobe the newly built it87.
Please notice the different numbers.
This is on Gigabyte X470 Aorus ULTRA GAMING board.

groeck, 2018-06-13:
I assume you mean the chip type with "numbers". The chip ID for IT8792E/IT8795E is 0x8733. The driver in the upstream kernel doesn't know that and reports IT8733E. The upstream driver also does not support IT8686E and thus does not report it. There is nothing unexpected in this log.

closed 2018-07-09 by groeck