groeck/it87 #94: Invalid numeric readings for Asus X470-Prime

rombert, 2018-06-14:
(moved from https://github.com/groeck/lm-sensors/issues/115)

After upgrading to git lm-senrors revision 1c48b19 and it87 revision b9a8431 I am able to get extended output for my motherboard. However, many of the readings seem invalid:

- min > max
- recorded values < min

Here's the full `sensors` output after loading the latest module version:

```
$ sensors
it8665-isa-0290
Adapter: ISA adapter
in0:          +0.79 V  (min =  +2.57 V, max =  +0.65 V)
in1:          +2.51 V  (min =  +1.21 V, max =  +2.03 V)
in2:          +2.01 V  (min =  +2.22 V, max =  +0.68 V)
in3:          +2.00 V  (min =  +0.48 V, max =  +1.18 V)
in4:          +0.03 V  (min =  +0.60 V, max =  +0.48 V)
in5:          +0.03 V  (min =  +0.57 V, max =  +2.04 V)
in6:          +0.03 V  (min =  +2.40 V, max =  +1.70 V)
3VSB:         +3.33 V  (min =  +4.40 V, max =  +4.64 V)
Vbat:         +3.21 V  
+3.3V:        +3.33 V  
fan1:         481 RPM  (min =   10 RPM)
fan2:         657 RPM  (min =   63 RPM)
fan3:         666 RPM  (min =   11 RPM)
fan4:         682 RPM  (min =   -1 RPM)
fan5:           0 RPM  (min =   -1 RPM)
fan6:           0 RPM  (min =   -1 RPM)
temp1:        +41.0°C  (low  = +115.0°C, high = -126.0°C)
temp2:        +39.0°C  (low  = -97.0°C, high = -32.0°C)  sensor = thermistor
temp3:        +42.0°C  (low  = +107.0°C, high = +43.0°C)  sensor = thermistor
temp4:        +42.0°C  (low  =  +4.0°C, high = -74.0°C)  sensor = thermistor
temp5:        +42.0°C  (low  =  -5.0°C, high = +127.0°C)  sensor = thermistor
temp6:        +42.0°C  (low  = +20.0°C, high = -69.0°C)  sensor = thermistor
intrusion0:  ALARM

k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +41.4°C  (high = +70.0°C)

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM
```

rombert, 2018-06-14:
@groeck  - you said that

> Did you try to initialize the limits ? The limits are often not or not correctly set by the BIOS.

How do I initialize the limits? I found no hint of that in the [module README](https://github.com/groeck/it87/blob/master/README)

groeck, 2018-06-14:
Now that is done with the "sensors" command. Add an entry for the chip in /etc/sensors3.conf or in /etc/sensors.d/<filename> and run "sudo sensors -s".

rombert, 2018-06-18:
What's the recommended way to get this data - spec sheets? Or checking what windows programs do?

groeck, 2018-06-18:
If you manage to get a specification from the board vendor, use it. Unfortunately, that is almost never the case. Most of the time people select a range (5 or 10 %) from the nominal voltage.

groeck, 2018-07-09:
Closing. Nothing to be done from driver perspective. Also please note driver removal notice in README file.

closed 2018-07-09 by groeck