groeck/it87 #96: it8768

stunpix, 2018-06-19:
I have Gigabyte GA-H110TN motherboard with it8768 and as expected it wasn't recognized:

```
Probing for Super-I/O at 0x2e/0x2f
...
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8768
    (logical device 4 has address 0x290, could be sensors)
```

Forced to load module with closest id `0x8771` and sensors output is following:

```
it8771-isa-0a40
Adapter: ISA adapter
in0:          +0.41 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.02 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.06 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +0.01 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +1.08 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.22 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.38 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.14 V  
fan1:        1214 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
temp1:        +38.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +48.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +35.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = Intel PECI
intrusion0:  ALARM
``` 

Looks ok?

BTW, I have it8772, it878(1,2,3) specification pdfs. Datasheets for them are stated  in readme as "not publicly available". Will they be helpful for project?

stunpix, 2018-06-19:
Here are dumps

```
# sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 28 02 09 00 00 b7 0c 22 00 80 40 88 80 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 0a 10 0a 00 0a 20 00 00 00 00 00 00 00 00 00 00 
70: 00 00 20 00 00 04 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 c0 00 00 00 00 
c0: 01 b3 0c 62 00 00 00 00 01 32 0c 22 00 00 00 00 
d0: c9 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 00 00 27 27 27 27 27 02 7e 
f0: 00 00 00 00 00 00 12 00 00 00 00 40 00 ff 00 0f 
```
and another one:

```
# sudo isadump 0x295 0x296

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
10: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
20: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
30: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
40: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
50: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
60: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
70: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
90: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
```

groeck, 2018-06-19:
> I have Gigabyte GA-H110TN motherboard with it8768 and as expected it wasn't recognized:
>
> ```
> Probing for Super-I/O at 0x2e/0x2f
> ...
> Trying family `ITE'...                                      Yes [...]
No, the information is quite inconsistent. The above says address 0x290,
yet the address displayed here is from 0x0a40. Something is wrong.

> BTW, I have it8772, it878(1,2,3) specification pdfs. Datasheets for them are stated  in readme as "not publicly available". Will they be helpful for project?
Those chips are already supported.

Guenter

groeck, 2018-06-19:
> Here are dumps
> 
> ```
> # sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
> 
>      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
> 00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
> 10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 20: 86 28 02 09 00 00 b7 0c 22 00 80 40 88 80 00 00 

That looks like an 8628, not 8768 ?

> 30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 60: 0a 10 0a 00 0a 20 00 00 00 00 00 00 [...]
You are looking for 0xa45/0xa46.

Guenter

stunpix, 2018-06-19:
> That looks like an 8628, not 8768 ?

Something strange is happening. Just checked BIOS - it reports [IT8768SEC](https://i.imgur.com/bbTnh6c.jpg), but when I opened my pc case and checked chip with my eyes – it's marked as [IT8628E](https://i.imgur.com/BmHKM0I.jpg). sensors-detect agrees with bios, but dump is not. Really weird.

> You are looking for 0xa45/0xa46.

Ok, here is another dump:
```
# sudo isadump 0xa45 0xa46

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 19 10 01 00 ff ff 18 20 ff 98 60 00 30 33 ff ff 
10: ff ff ff 77 c7 82 80 82 02 ff ff ff ff ff ff ff 
20: 2c a8 ac a7 01 5a 66 8d 83 25 30 23 5e 21 21 7d 
30: ff 00 ff 00 ff 00 ff 00 ff 00 ff 00 ff 00 ff 00 
40: 7f 7f 7f 7f 7f 7f 5f 40 ad 6a d4 00 ff ff 00 00 
50: ff d8 7f 7f 7f 40 00 00 90 50 1b 12 65 00 00 00 
60: 00 14 41 46 0c 03 00 05 00 14 3c 46 1c 03 00 ff 
70: 00 14 41 46 1c 03 00 3f 00 00 73 00 00 00 0f 82 
80: ff ff ff ff ff ff ff ff 02 30 01 02 01 8e e0 fe 
90: ff 00 00 00 ff 00 00 00 40 97 00 00 40 ac c1 14 
a0: 00 00 96 00 00 0a 0f 80 7f 7f 96 80 00 00 0f 80 
b0: d3 08 73 60 7f 00 7f 00 7f 00 3c 3c 3c 3c 3c 3c 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
```

groeck, 2018-07-09:
Closing. My contact to the chip vendor dried up, and I can no longer get information about new chips. Also please see driver removal notice in README file.

closed 2018-07-09 by groeck