groeck/it87 #98: IT8721F/IT8758E has min larger than max with Acer Aspire R3700

RalphCorderoy, 2018-06-20:
Arch Linux with package 'lm_sensors' 3.4.0-4 recently upgraded the
'linux' package from 4.16.13-2 -> 4.17.2-1 and sensors(1) then produced
lots more entries.  It now gives

      $ sensors
      it8721-isa-0a10
      Adapter: ISA adapter
      in0:          +2.93 V  (min =  +2.51 V, max =  +3.00 V)
    → in1:          +2.93 V  (min =  +2.26 V, max =  +0.88 V)  ALARM
    → in2:          +1.14 V  (min =  +1.98 V, max =  +0.71 V)  ALARM
    → +3.3V:        +3.34 V  (min =  +5.78 V, max =  +5.54 V)  ALARM
      in4:          +1.04 V  (min =  +0.19 V, max =  +1.32 V)
      in5:          +2.22 V  (min =  +2.22 V, max =  +2.34 V)  ALARM
      in6:          +2.77 V  (min =  +1.42 V, max =  +3.06 V)
    → 3VSB:         +3.29 V  (min =  +6.00 V, max =  +0.94 V)  ALARM
      Vbat:         +3.22 V
      fan1:        5037 RPM  (min =   20 RPM)
      fan2:           0 RPM  (min =   15 RPM)  ALARM
      fan3:           0 RPM  (min =   59 RPM)  ALARM
    → temp1:        +65.0°C  (low  = +124.0°C, high = -81.0°C)  ALARM  sensor = thermal diode
      temp2:        +30.0°C  (low  =  -6.0°C, high = -71.0°C)  ALARM  sensor = thermistor
    → temp3:        +43.0°C  (low  = +48.0°C, high = -71.0°C)  ALARM  sensor = thermal diode
      intrusion0:  ALARM

      acpitz-virtual-0
      Adapter: Virtual device
      temp1:        +65.0°C  (crit = +127.0°C)
      temp2:        +35.0°C  (crit = +127.0°C)

      coretemp-isa-0000
      Adapter: ISA adapter
      Core 0:       +32.0°C  (crit = +100.0°C)
      Core 1:       +32.0°C  (crit = +100.0°C)
      $

I marked the ones that look odd due to 'min > max'.

The sysfs files match, e.g.

    $ grep ^ /sys/class/hwmon/hwmon2/in1_*
    /sys/class/hwmon/hwmon2/in1_alarm:1
    /sys/class/hwmon/hwmon2/in1_beep:0
    /sys/class/hwmon/hwmon2/in1_input:2928
    /sys/class/hwmon/hwmon2/in1_max:876
    /sys/class/hwmon/hwmon2/in1_min:2256

Here's the information that I think you request.

    $ sudo -i sensors-detect
    ...
    Probing for Super-I/O at 0x2e/0x2f
    ...
    Trying family `ITE'...                                      Yes
    Found `ITE IT8721F/IT8758E Super IO Sensors'                Success!
        (address 0xa10, driver `it87')
    ...
    $
    $
    $ sudo -i isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
    WARNING! Running this program can cause system crashes, data loss and worse!
    I will probe address register 0x2e and data register 0x2f.
    Probing bank 7 using bank register 0x07.
    Continue? [Y/n]
         0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
    00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00
    10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    20: 87 21 01 00 00 00 0c 01 00 01 00 00 03 00 00 00
    30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    60: 00 00 0a 20 0a 30 00 00 00 00 00 00 00 00 00 00
    70: 00 00 00 00 00 04 00 00 00 00 00 00 00 00 00 00
    80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    b0: 00 00 00 00 00 00 00 00 20 00 00 01 00 00 00 00
    c0: 01 08 01 01 00 00 00 00 01 08 01 01 00 00 00 00
    d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    e0: 00 00 00 00 00 00 00 00 00 07 07 07 c4 3e 3e 0c
    f0: 08 00 00 00 00 28 18 00 00 00 00 00 fb 00 00 00
    $
    $ sudo -i isadump 0xa15 0xa16
    WARNING! Running this program can cause system crashes, data loss and worse!
    I will probe address register 0xa15 and data register 0xa16.
    Continue? [Y/n]
         0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
    00: 19 16 ae 07 00 00 00 00 00 80 58 0f 07 7d ff ff
    10: 36 c1 6d 77 c7 80 00 00 00 ff ff 80 ae 2c 2c 2c
    20: f4 f4 5f 8b 57 b9 e7 89 86 43 1d 2e 80 30 33 33
    30: fa d1 49 bc 3b a5 e7 f1 6e 10 c3 b9 ff 76 27 fa
    40: af 7c b9 fa b9 30 5f 40 ad 6a d4 00 ff ff ff ff
    50: ff 15 7f 7f 7f 40 e9 00 90 00 3b 12 e0 00 00 00
    60: 00 35 7f 55 a0 02 00 ff 7f 7f 7f 80 00 00 00 ff
    70: 7f 7f 7f 80 00 00 00 ff ff ff ff ff ff ff ff ff
    80: 00 00 00 00 00 00 00 00 00 00 00 00 00 95 02 2b
    90: 3c 78 23 00 2c f8 24 00 ff ff ff ff ff ff ff ff
    a0: f7 c6 c6 c6 c6 c6 c6 ff ff ff ff ff ff ff ff ff
    b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
    c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
    d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
    e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
    f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
    $

The PC's details, mainly for those that might Google for them.

    $ sudo -i dmidecode
    ...
        Vendor: American Megatrends Inc.
        Version: P01-A1L
        Release Date: 07/30/2010
        BIOS Revision: 8.16
    ...
        Manufacturer: Acer
        Product Name: Aspire R3700
    ...
        Manufacturer: Acer
        Product Name: TDPS05 R3700

groeck, 2018-06-20:
Limits are almost never set correctly by the BIOS. Did you try to configure and set limits ?

RalphCorderoy, 2018-06-21:
I didn't realise the limits came from the BIOS, I assumed they were tied
to the IT8721.

The BIOS has no means to alter min/max and doesn't even display them.
It displays

    CPU temperature     67°C      
    System temperature  28°C      
    CPU fan speed       5400      
    CPU core            1.140 V   
    +1.05 V             1.044 V   
    +3.30 V             3.326 V   
    +5.00 V             5.006 V
    5VSB                5.036 V
    VBAT                3.384 V
    Smart fan enabled

Trying to marry those with sensors(1) output from soon after gives:

                                  it8721-isa-0a10
                                  Adapter: ISA adapter
                                  in0:          +2.93 V  (min =  +2.51 V, max =  +3.00 V)
                                  in1:          +2.93 V  (min =  +2.26 V, max =  +0.88 V)  ALARM
    CPU core            1.140 V   in2:          +1.14 V  (min =  +1.98 V, max =  +0.71 V)  ALARM
    +3.30 V             3.326 V   +3.3V:        +3.34 V  (min =  +5.78 V, max =  +5.54 V)  ALARM
    +1.05 V             1.044 V   in4:          +1.04 V  (min =  +0.19 V, max =  +1.32 V)
                                  in5:          +2.22 V  (min =  +2.22 V, max =  +2.34 V)  ALARM
                                  in6:          +2.77 V  (min =  +1.42 V, max =  +3.06 V)
                                  3VSB:         +3.29 V  (min =  +6.00 V, max =  +0.94 V)  ALARM
    VBAT                3.384 V   Vbat:         +3.38 V  
    CPU fan speed       5400      fan1:        4821 RPM  (min =   20 RPM)
                                  fan2:           0 RPM  (min =   15 RPM)  ALARM
                                  fan3:           0 RPM  (min =   59 RPM)  ALARM
    CPU temperature     67°C      temp1:        +64.0°C  (low  = +124.0°C, high = -81.0°C)  ALARM  sensor = thermal diode
    System temperature  28°C      temp2:        +28.0°C  (low  =  -6.0°C, high = -71.0°C)  ALARM  sensor = thermistor
                                  temp3:        +42.0°C  (low  = +48.0°C, high = -71.0°C)  ALARM  sensor = thermal diode
                                  intrusion0:  ALARM

With these unmatched:

    +5.00 V             5.006 V
    5VSB                5.036 V

I'll try and see if I can `compute' these from some of those above.
Can sensors(1) output with greater precision?  The BIOS, and 'sensors
-u' produce '%.3f' for voltages.

RalphCorderoy, 2018-06-21:
The attached conf file gives

    it8721-isa-0a10
    Adapter: ISA adapter
    +5.00V:       +4.92 V  (min =  +4.86 V, max =  +5.14 V)
    5VSB:         +4.92 V  (min =  +4.86 V, max =  +5.14 V)
    cpu:          +1.14 V  (min =  +1.13 V, max =  +1.18 V)
    +3.30V:       +3.34 V  (min =  +3.24 V, max =  +3.36 V)
    +1.05V:       +1.04 V  (min =  +1.03 V, max =  +1.07 V)
    in5:          +2.22 V  (min =  +2.17 V, max =  +2.27 V)
    in6:          +2.77 V  (min =  +2.71 V, max =  +2.82 V)
    3VSB:         +3.29 V  (min =  +3.24 V, max =  +3.36 V)
    VBAT:         +3.38 V  
    cpu-fan:     5400 RPM  (min = 4787 RPM)  ALARM
    cpu-temp:     +65.0°C  (low  = +61.0°C, high = +67.0°C)  ALARM  sensor = thermal diode
    system-temp:  +28.0°C  (low  = +27.0°C, high = +29.0°C)  sensor = thermistor
    temp3:        +46.0°C  (low  = +40.0°C, high = +44.0°C)  ALARM  sensor = thermal diode

    acpitz-virtual-0
    Adapter: Virtual device
    cpu-temp:     +65.0°C  (crit = +127.0°C)
    system-temp:  +33.0°C  (crit = +127.0°C)

    coretemp-isa-0000
    Adapter: ISA adapter
    Core 0:       +32.0°C  (crit = +100.0°C)
    Core 1:       +33.0°C  (crit = +100.0°C)

There's still a few unknowns.  Any suggestions welcome.  Thanks for the hint about 'sensors -s' to set the min., etc.

[acer-aspire-r3700.txt](https://github.com/groeck/it87/files/2123425/acer-aspire-r3700.txt)

groeck, 2018-06-21:
For the fans and temperatures you should set reasonable limits, such as 500 rpm for the fan and 0/+80 for the temperatures. You can see raw values with "sensors -u".

RalphCorderoy, 2018-06-22:
Thanks.  I'm planning on expanding the allowable min/max as I see them alarm during normal behaviour so they're highlighting future anomalies.  Yes, I agree about the -u, I was using that when matching up the BIOS's results.  Thanks again for your help.

closed 2018-06-22 by RalphCorderoy