groeck/it87 #99: it87 does not load on ASUS ROG STRIX X399-E

aceman444, 2018-06-30:
On the ASUS ROG STRIX X399-E board (TR4 Threadripper socket) I get no voltage and fan speed sensors.

sensors-detect finds this:
Trying family `ITE'...                                      Yes                                                                                                                                                                              
Found unknown chip with ID 0x8665                                                                                                                                                                                                            
    (logical device 4 has address 0x290, could be sensors)     

Looking at issues like issue #80 it seems to me with sensors on this address and ID the it87 module should be loaded, but isn't. I get no "it8665-isa-0290" output from 'sensors'.
When loading the module, I get 'modprobe: ERROR: could not insert 'it87': No such device' .

I have the latest 0601 BIOS installed from ASUS. lm-sensors package is 3.4.0.
Only k10temp and amdgpu modules seem to work fine.

aceman444, 2018-06-30:
I have even enabled 'ISA bus support on modern systems' option in the kernel as the output in issue 80 (and 'lspci') shows the sensors to reside on "ISA adapter". Otherwise I run with ISA disabled as there are only PCI express slots on the MB.

What else may need to be enabled in kernel?

groeck, 2018-06-30:
Please install and run the driver from here instead of the upstream driver.

groeck, 2018-06-30:
"Will you please upstream the changes"

Yeah, sure, only no one is really willing to help, or even remotely takes into account that I am not getting paid for this work and may have a limited amount of time available. It would take the better part of a month to prepare patches to take it upstream, and then I would need people to review those patches, which would be a challenge.

Note that I would be more than happy to review and accept patches from others into the upstream kernel to bring the drivers at par. It doesn't have to be me doing that work. Would you be willing to sign up for that ?

The motherboard may have a fan multiplexer and/or an undocumented ASIC to connect to the additional fans. Only the motherboard vendor knows, and does not share this information. If you can convince the vendor to share the necessary information, I'll be more than happy to at least try to add support for it. Someone would also have to donate a board and CPU, though. I would also be more than happy to review and integrate the necessary patches from others.

The driver does not need ISA bus support enabled. The term used in the sensors command is historic. The chip uses LPC and optionally memory mapped IO to communicate with the CPU.

aceman444, 2018-06-30:
Well, I can't prepare or review patches for the driver, I can only test it for you :)
But if you already have the functionality done (for the it8665 chip), why would it need so long to get upstreamed? Do you need to rewrite the patches to have proper coding style and documentation to be accepted in kernel?

groeck, 2018-06-30:
Upstream rule is one patch per logical change, which means we'll need maybe 50 separate patches to upstream the code.

aceman444, 2018-06-30:
But surely you already did those changes here in git and they are tracked as patches.
It is a pity the improvements rot here and boards that are a year old still do not have sensors supported in Linux. I also only got here to find the code by accident.

groeck, 2018-07-01:
Doing a change here does not need to be reviewed. I can put as much junk here as I want. The upstream kernel is just a little more stringent. The lack of reviewers caused the drivers to deviate in the first place. Now it is very difficult and work intensive for the upstream kernel to catch up.
Anyway, do you prefer for me to mark this driver private ? Just let me know, and I'll be happy to comply.

groeck, 2018-07-09:
Closing. Please note driver removal notice in README file.

closed 2018-07-09 by groeck

aceman444, 2018-07-18:
Of course I don't want that the driver vanishes and all the work is lost. Linux needs working and up-to-date drivers. Sadly I can't help with maintaining this code, I do other projects.